Pro/open62541-arduino #4: I followed your "JTAG Debugging ESP32" instructions and it destroyed my developpent board

SDS-Emb, 2018-08-30:
I followed exactly your instruction for JTAG Debugging ESP32 and it destroyed my developpement board with platformIO instructions.

First it stopped to work and beeing detected by the PC, and after a retry the CP2104 component burnt !

NB: think however that the target is still programmable via JTAG controller (if it is correctly wired)

Pro, 2018-08-30:
Thanks for the info, but as always with microcontroller programming: You really have to know what you are doing. There can be many reasons why your board broke. The only think I know for sure is that the procedure worked for me. Note that I did not use platform.io at all...

closed 2018-08-30 by Pro

SDS-Emb, 2018-08-30:
Very strange I verified many times the wiring before connecting with the two USB cables. 
One USB between PC and the Olimex tool, and the other was between PC and Adafruit, isn't it ?

SDS-Emb, 2018-08-31:
@Pro I think the wiring cause problems as soon as I plug the 3V with the probe the card shutdown. Know that the CP2104 burnt only after many usages. It was not from the first time (this might be an overvoltage anywhere)