Pro/open62541-arduino #5: The wiring in the tutorial is all reversed

SDS-Emb, 2018-09-03:
The wiring in the tutorial is all reversed, this is why the usb chip burnt !

Look at the schema vs the photo below, they don't correspond. On the schema you've wired a MALE connector to the HUZZAH32, so when we see it and wire our FEMALE OLIMEX ARM OCD H connector the result is that all is reversed 
(schema)
[image]

(photo)
[image]

Please correct your tutorial. 

For my case it still don't work, but the problem can be now from the probe: when the Adafruit burnt it may have damaged the probe also.

Pro, 2018-09-03:
Thank you for the update!!
The schematics are correct, since you normally draw the socket (20 Pin Interface) in schematics, and not the plug.

So what I will do is to make it more clear that the schematics are showing the socket part.

SDS-Emb, 2018-09-03:
[image]

This does not inform us how to connect in the connector, it's the schema which inform us.

Pro, 2018-09-03:
I updated the wiki page so it should be a bit clearer now, anyways it makes sense to read through the whole document before starting to work with it. Especially make sure that your wiring matches the photos

SDS-Emb, 2018-09-03:
@Pro ok it's better now but you should have drawn the correct female socket instead of the note saying that we have to mirror.

Pro, 2018-09-03:
Please have a look at this picture here:
https://i.stack.imgur.com/7vwFx.png

The socket is not the female one (just to be clear: female is the plug with the many holes). It is the male socket (as shown in the picture).

SDS-Emb, 2018-09-03:
@Pro I wanted to say: "ok it's better now but you should have drawn the correct female connector instead of the note saying that we have to mirror."

and off course as you can see in your photos this is the problem I told you: this schema drawn is for a male JTAG connector while there only the femal plug in our setup handling. 

note also that the ancient schema had also the poka yoke of the female connector while it was a male connector

Pro, 2018-09-03:
It is exactly how I wanted to have it drawn. The schematic uses a socket and not plug, how it is in nearly every other drawing which you will find in the internet. It is common sense and that's how it should be.

I will not change the drawing to use the mirrored male plug since that is misleading.

Feel free to open a new issue if you have other problems, but please accept the fact that I made a decision how to document it and feel free to write your own documentation if you feel the urge to draw it mirrored.

I will ignore any further comments which are related to the drawing of the socket, since I have other open source projects to manage.

piotrkundu, 2019-10-24:
@Pro I'm a bit late to the party, but I think @SDS-Emb has a point. And I think adding @SDS-Emb pictures of a female being connected to a breadboard is even more confusing. Now you assume it is the female showing. It might be obvious for some, but the difference between male and female is just to characters.