Pro/open62541-arduino #6: Stop closing open issues or at least answer to them !!

SDS-Emb, 2018-09-03:
Your documentaiton si unclear so correct it. This is very impolite to close the issue while your tutorial have made my 18€ board burned and probably my 55€ Olimex probe damaged as well !

You're not cooperative ! You just make some rash conclusions, without any discussion, and close the subject, with not any other answer.

Pro, 2018-09-03:
@SDS-Emb since it is my repository I can close the issues whenever I want, especially if I think an issue is closed!
I am very polite since I am still helping you with your issue (it's not my issue by the way). Also I do not have to put any information online! So the only impolite stuff happening within this repository is your post here.

I am sorry that you burnt your board(s), but if you are not able to read through the documentation carefully it won't be the last board you burnt. And please read through the Code of Conduct before you post such a message again!
https://github.com/open62541/open62541/blob/master/CODE_OF_CONDUCT.md

And also, you should not create a new issue for every comment you are writing. I will also get your comments on a closed issue. That's not how you should behave in an open source community.

closed 2018-09-03 by Pro

SDS-Emb, 2018-09-03:
@Pro 

> I will also get your comments on a closed issue.

Look at here you closed the issue and stopped to answer. 
https://github.com/Pro/open62541-arduino/issues/4

> it's not my issue by the way

Wrong, it's a documentation issue (your documentation), and if the documentation followed burns board, it is a major documentation issue. 

> but if you are not able to read through the documentation carefully it won't be the last board you burnt

I've read it carefully and followed strictly your schema which was very misleading

Pro, 2018-09-03:
It's obvious that you did NOT follow the insturctions! There was the clear note that it is the socket. And if you compare the schematics with the photo you would have seen that it is correct.

I will not start a discussion here how to handle open source projects, since it's off-topic. I'm an open source developer since many years.

So please only post issues if it is related to the project itself, otherwise I will ignore, or if necessary delete unrelated posts.