Pro/open62541-arduino #10: Can't run example

mmuecke, 2018-12-16:
I tried to create the example.
If I follow the instructions I get these messages:
mkdir build && cd build
cmake -DWIFI_SSID="SOME_SSID" -DWIFI_PWD="SOME_PWD" ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Arduino SDK version 1.0.5: /usr/share/arduino
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- Architectures included in amalgamation: freertosLWIP
-- The selected architecture is: freertosLWIP
-- Found Sphinx: /usr/bin/sphinx-build  
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Generating arduino_opcua
-- Generating featheresp32_WiFi for library WiFi
CMake Error at cmake/Platform/Arduino.cmake:896 (MESSAGE):
  Value for .recipe.cpp.o.pattern not defined
Call Stack (most recent call first):
  cmake/Platform/Arduino.cmake:1139 (get_recipe_flags)
  cmake/Platform/Arduino.cmake:1227 (setup_arduino_library)
  cmake/Platform/Arduino.cmake:538 (setup_arduino_libraries)
  CMakeLists.txt:84 (generate_arduino_firmware)

mmuecke, 2018-12-16:
I have progressed by downloading the latest version from the Arduino website. But I'm getting this error now.

[ 96%] Building C object CMakeFiles/arduino_opcua.dir/lib/open62541/open62541.c.obj
/home/mmuecke/open62541-arduino/build/lib/open62541/open62541.c:8721:9: note: #pragma message: No native IEEE 754 format detected. Use slow generic encoding.
 #pragma message "No native IEEE 754 format detected. Use slow generic encoding."
         ^
[100%] Linking CXX executable arduino_opcua.elf
xtensa-esp32-elf-g++: error: {archive_file_path}: No such file or directory
CMakeFiles/arduino_opcua.dir/build.make:201: recipe for target 'arduino_opcua.elf' failed
make[2]: *** [arduino_opcua.elf] Error 1
CMakeFiles/Makefile2:279: recipe for target 'CMakeFiles/arduino_opcua.dir/all' failed
make[1]: *** [CMakeFiles/arduino_opcua.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

mmuecke, 2018-12-19:
I'm sorry the error description was so brief. I tried it again. My procedure:
1. VM Ware Ubuntu 18.04 created from ISO
2. everything prepared with the following commands:
```
# Install packages
sudo apt-get install git python python-serial python-pip avr-libc avrdude binutils-avr gcc gcc-7 gcc-avr libftdi1 extra-xdg-menus libftdi1 libjna-java libjna-jni librxtx-java libusb-0.1-4 cmake 
pip install six
sudo apt-get installca-certificates-java default-jre default-jre-headless extra-xdg-menus fonts-dejavu-extra java-common libasan4 libatk-wrapper-java libatk-wrapper-java-jni libatomic1 libc-dev-bin libc6-dev libcilkrts5 libftdi1 libgcc-7-dev libgif7 libitm1 libjna-java libjna-jni liblsan0 libmpx2 libquadmath0 librxtx-java libtsan0 libubsan0 libusb-0.1-4 linux-libc-dev manpages-dev openjdk-11-jre openjdk-11-jre-headless

git clone https://github.com/Pro/open62541-arduino.git
cd open62541-arduino
ROOT_DIR=$PWD
git submodule update --init --recursive

# Get arduino sdk
wget https://downloads.arduino.cc/arduino-1.8.8-linux64.tar.xz
tar xf arduino-1.8.8-linux64.tar.xz
rm -r arduino-1.8.8-linux64.tar.xz

cd arduino-1.8.8
ARDUINO_SDK_PATH=$PWD
cd hardware
# Additionally get the ESP toolchain
mkdir espressif && cd espressif

git clone --depth 1 https://github.com/espressif/arduino-esp32 esp32

cd esp32

git submodule update --init --recursive
cd tools && python get.py
cd ..

chmod +x tools/esptool/esptool.py

# Increase the stack size
sed -i 's/xTaskCreatePinnedToCore(loopTask, "loopTask", 8192, NULL, 1, NULL, ARDUINO_RUNNING_CORE);/xTaskCreatePinnedToCore(loopTask, "loopTask", 16384, NULL, 1, NULL, ARDUINO_RUNNING_CORE);/g' cores/esp32/main.cpp

# Install arduino SDK
cd $ROOT_DIR
sudo mv arduino-1.8.8 /opt
cd /opt/arduino-1.8.8
ARDUINO_SDK_PATH=$PWD
sudo chmod +x install.sh
./install.sh

cd $ROOT_DIR
```

3. start build

```
mkdir build && cd build
cmake -DWIFI_SSID="SOME_SSID" -DWIFI_PWD="SOME_PWD" ..
make -j
if [ $? -ne 0 ] ; then exit 1 ; fi
```

Same result as before, the error still exists:
`xtensa-esp32-elf-g++: error: {archive_file_path}: No such file or directory`

menshen-tsai, 2018-12-29:
Hi!!

I solved the problem by doing the following:

In ~/open62541-arduino/arduino-1.8.6/hardware/espressif/esp32/platform.txt

Change 

## Create archives

To
## Create archives
archive_file_path={build.path}/{archive_file}

kusumy, 2019-10-18:
> Hi!!
> 
> I solved the problem by doing the following:
> 
> In ~/open62541-arduino/arduino-1.8.6/hardware/espressif/esp32/platform.txt
> 
> Change
> 
> `## Create archives`
> 
> To
> `## Create archives`
> archive_file_path={build.path}/{archive_file}

on my platform.txt there is no line ## create archives, 
so i make a new line :

## Create archives
archive_file_path={build.path}/{archive_file}

But there is still error messages:
xtensa-esp32-elf-g++: error: {archive_file_path}: No such file or directory

is there something I've missed?

kusumy, 2019-10-18:
I've just realized that i don't have arduino-1.8.xx folder in my open62541-arduino folder.
My arduino is installed on /opt/arduino-1.8.xx

is there something I've missed?

kusumy, 2019-10-18:
> I'm sorry the error description was so brief. I tried it again. My procedure:
> 
>     1. VM Ware Ubuntu 18.04 created from ISO
> 
>     2. everything prepared with the following commands:
> 
> 
> ```
> # Install packages
> sudo apt-get install git python python-serial python-pip avr-libc avrdude binutils-avr gcc gcc-7 gcc-avr libftdi1 extra-xdg-menus libftdi1 libjna-java libjna-jni librxtx-java libusb-0.1-4 cmake 
> pip install six
> sudo apt-get installca-certificates-java default-jre default-jre-headless extra-xdg-menus fonts-dejavu-extra java-common libasan4 libatk-wrapper-java libatk-wrapper-java-jni libatomic1 libc-dev-bin libc6-dev libcilkrts5 libftdi1 libgcc-7-dev libgif7 libitm1 libjna-java libjna-jni liblsan0 libmpx2 libquadmath0 librxtx-java libtsan0 libubsan0 libusb-0.1-4 linux-libc-dev manpages-dev openjdk-11-jre openjdk-11-jre-headless
> 
> git clone https://github.com/Pro/open62541-arduino.git
> cd open62541-arduino
> ROOT_DIR=$PWD
> git submodule update --init --recursive
> 
> # Get arduino sdk
> wget https://downloads.arduino.cc/arduino-1.8.8-linux64.tar.xz
> tar xf arduino-1.8.8-linux64.tar.xz
> rm -r arduino-1.8.8-linux64.tar.xz
> 
> cd arduino-1.8.8
> ARDUINO_SDK_PATH=$PWD
> cd hardware
> # Additionally get the ESP toolchain
> mkdir espressif && cd espressif
> 
> git clone --depth 1 https://github.com/espressif/arduino-esp32 esp32
> 
> cd esp32
> 
> git submodule update --init --recursive
> cd tools && python get.py
> cd ..
> 
> chmod +x tools/esptool/esptool.py
> 
> # Increase the stack size
> sed -i 's/xTaskCreatePinnedToCore(loopTask, "loopTask", 8192, NULL, 1, NULL, ARDUINO_RUNNING_CORE);/xTaskCreatePinnedToCore(loopTask, "loopTask", 16384, NULL, 1, NULL, ARDUINO_RUNNING_CORE);/g' cores/esp32/main.cpp
> 
> # Install arduino SDK
> cd $ROOT_DIR
> sudo mv arduino-1.8.8 /opt
> cd /opt/arduino-1.8.8
> ARDUINO_SDK_PATH=$PWD
> sudo chmod +x install.sh
> ./install.sh
> 
> cd $ROOT_DIR
> ```
> 
>     1. start build
> 
> 
> ```
> mkdir build && cd build
> cmake -DWIFI_SSID="SOME_SSID" -DWIFI_PWD="SOME_PWD" ..
> make -j
> if [ $? -ne 0 ] ; then exit 1 ; fi
> ```
> 
> Same result as before, the error still exists:
> `xtensa-esp32-elf-g++: error: {archive_file_path}: No such file or directory`

have you solved this problem?

felipecvb, 2019-10-24:
Hi!
I was having this issue too. After I added that new line in platform.txt it fixed but now other issue show up:

```
[100%] Linking CXX executable arduino_opcua.elf
Generating EEP image
usage: gen_esp32part.py [-h] [--flash-size [{1MB,2MB,4MB,8MB,16MB}]]
                        [--disable-md5sum] [--no-verify] [--verify] [--quiet]
                        [--offset OFFSET] [--secure]
                        input [output]
gen_esp32part.py: error: argument input: can't open '{build.path}/partitions.csv': [Errno 2] No such file or directory: '{build.path}/partitions.csv'
CMakeFiles/arduino_opcua.dir/build.make:201: recipe for target 'arduino_opcua.elf' failed
make[2]: *** [arduino_opcua.elf] Error 2
make[2]: *** Deleting file 'arduino_opcua.elf'
CMakeFiles/Makefile2:279: recipe for target 'CMakeFiles/arduino_opcua.dir/all' failed
make[1]: *** [CMakeFiles/arduino_opcua.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

kusumy, 2019-10-25:
> Hi!
> I was having this issue too. After I added that new line in platform.txt it fixed but now other issue show up:
> 
> ```
> [100%] Linking CXX executable arduino_opcua.elf
> Generating EEP image
> usage: gen_esp32part.py [-h] [--flash-size [{1MB,2MB,4MB,8MB,16MB}]]
>                         [--disable-md5sum] [--no-verify] [--verify] [--quiet]
>                         [--offset OFFSET] [--secure]
>                         input [output]
> gen_esp32part.py: error: argument input: can't open '{build.path}/partitions.csv': [Errno 2] No such file or directory: '{build.path}/partitions.csv'
> CMakeFiles/arduino_opcua.dir/build.make:201: recipe for target 'arduino_opcua.elf' failed
> make[2]: *** [arduino_opcua.elf] Error 2
> make[2]: *** Deleting file 'arduino_opcua.elf'
> CMakeFiles/Makefile2:279: recipe for target 'CMakeFiles/arduino_opcua.dir/all' failed
> make[1]: *** [CMakeFiles/arduino_opcua.dir/all] Error 2
> Makefile:151: recipe for target 'all' failed
> make: *** [all] Error 2
> ```
> 
> I tried to replace the build.path the location of the folder but still didn't work.

same issue wtih me ...

Pro, 2019-10-25:
Unfortunately I did stop supporting this project due to two reasons:

1. The arduino-cmake modifications I did are not integrated in the main repo. Keeping that fork up-to-date is too much effort. See also https://github.com/arduino-cmake/arduino-cmake/pull/74

2. In general it's better to use the esp32 framework directly, instead of using arduino. I'm currently in the process of creating an example repo for open62541 + ESP32 without arduino. See also: https://github.com/Pro/open62541-esp32 (more code will follow the coming days)

kusumy, 2019-10-25:
> 
> 
> Unfortunately I did stop supporting this project due to two reasons:
> 
>     1. The arduino-cmake modifications I did are not integrated in the main repo. Keeping that fork up-to-date is too much effort. See also [arduino-cmake/arduino-cmake#74](https://github.com/arduino-cmake/arduino-cmake/pull/74)
> 
>     2. In general it's better to use the esp32 framework directly, instead of using arduino. I'm currently in the process of creating an example repo for open62541 + ESP32 without arduino. See also: https://github.com/Pro/open62541-esp32 (more code will follow the coming days)

Please announce if the code already available ... :)

Pro, 2019-11-29:
@kusumy and others:

The example for open62541 on an ESP32 using the ESP IDF Framework is now available here:

https://github.com/Pro/open62541-esp32