Pro/open62541-arduino #12: Subscription stops with decreasing stack

sammer-h, 2019-02-10:


Pro, 2019-02-25:
We are using the stack on an ESP32 and there we do not have memory issues.

Maybe there's a free missing in your code?
We are testing the full stack for memory leaks which means there shouldn't be any in the core.

Can you post your code example/main which you are using.

sammer-h, 2019-02-26:
It's nothing special at all - I basically took your code for Arduino.
But there seems to be the same or at least a similar problem with other freertos-lwip platforms.
After writing this post I found the following issue:
https://github.com/open62541/open62541/issues/2227
Near the end of the issue (Jan 28) [FishEatingCat](/FishEatingCat) describes the problem.

```
...
xTaskCreate(&opcua_task, "opcua_task", 1024 * 8, NULL, 1, NULL);
...
void opcua_task(void *pvParameter) {
   	ESP_LOGI(TAG, "Fire up OPC UA Server.");

	config = UA_ServerConfig_new_customBuffer(4840, NULL, 8192, 8192);
	UA_ServerConfig_set_customHostname(config, UA_STRING("espressif"));
	server = UA_Server_new(config);
	UA_StatusCode retVal = UA_Server_run_startup(server);

	while (1) 	{
		UA_Server_run_iterate(server, true);

		struct timeval tv;
		time_t nowtime;

		gettimeofday(&tv, NULL);
		nowtime = tv.tv_sec;

		vTaskDelay(20 / portTICK_PERIOD_MS);
		if (nowtime - lastMemOutput > 5) {
			memUsage();
			lastMemOutput = nowtime;
		}
	}

    ESP_LOGI(TAG, "RAM left %d", esp_get_free_heap_size());

    ESP_LOGI(TAG, "Now going to stop the server.");
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);

    ESP_LOGI("OPC_TASK", "opcua_task going to return");
    vTaskDelete(NULL);
}
```

sammer-h, 2019-02-26:
[image]
<- console output of ESP32 when everything is working fine

[image]
<- console output of ESP32 if subscription stopps

[image]
<- corresponding log of UA Expert

kusumy, 2019-10-25:
hi @sammer-h, can you tell me how to build this repo?
I never succeed to build this repo.

Always have this error:

[ 98%] Linking CXX executable arduino_opcua.elf
Generating EEP image
usage: gen_esp32part.py [-h] [--flash-size [{1MB,2MB,4MB,8MB,16MB}]]
                        [--disable-md5sum] [--no-verify] [--verify] [--quiet]
                        [--offset OFFSET] [--secure]
                        input [output]
gen_esp32part.py: error: argument input: can't open '{build.path}/partitions.csv': [Errno 2] No such file or directory: '{build.path}/partitions.csv'
CMakeFiles/arduino_opcua.dir/build.make:201: recipe for target 'arduino_opcua.elf' failed
make[2]: *** [arduino_opcua.elf] Error 2
make[2]: *** Deleting file 'arduino_opcua.elf'
CMakeFiles/Makefile2:279: recipe for target 'CMakeFiles/arduino_opcua.dir/all' failed
make[1]: *** [CMakeFiles/arduino_opcua.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

dj-fiorex, 2021-02-10:
@kusumy Please see this #16  to be able to build successfully
@sammer-h  hello, did you resolve your issue? do you use wifi or eth on your project?