Pro/open62541-arduino #14: Possibility of using LAN vs WLAN

puchm, 2020-04-08:
Hi,

I was wondering whether it would be possible to use LAN instead of WLAN for this? I would like to use an ethernet cable to connect to the microcontroller. If adjustments to the code are needed for this I can try to do that as well, I would just appreciate some information on what I would have to change for that and whether it is actually possible.
I know that this project is no longer maintained but I wanted to try it anyways.

Thank you for your help!
-Moritz

Pro, 2020-04-08:
Note that this repo is not maintained anymore (as noted in the readme).
It should be possible to switch to LAN by changing the LWIP settings.

anyways I recommend to use the following repository (if you are using an ESP32 chip), which already supports LAN and WLAN:
https://github.com/Pro/open62541-esp32

puchm, 2020-04-14:
Thank you for your reply.
The whole ESP32 thing confused me a little bit. Are there Arduino boards that are ESP32 chips or why would you recommend to use the other repository?
I am trying to understand why you would link a project for what seems to me like a different microcontroller.

puchm, 2020-04-15:
Hi!
I have another follow-up question. Unfortunately using Arduino is a requirement for the project so using a different chip is sadly not an option.
I am trying to compile the code and I think I am almost there. Here's what I did:
- Cloned the repo (with --recursive flag)
- run `tools/travis/travis_linux_before_install.sh`
- set `ARDUINO_SDK_PATH` to the location of the `arduino-1.8.6` folder
- created build folder and entered it
- run `cmake -DWIFI_SSID="SOME_SSID" -DWIFI_PWD="SOME_PWD" -DUSE_WIFI=OFF ..` - I used the default values since I don't think they will get used anyways but if I leave them out it throws an error.
- run `make -j`

Then I got an error:
`/home/[...]/open62541-arduino/tools/travis/arduino-1.8.6/hardware/espressif/esp32/tools/sdk/include/soc/soc/cpu.h:36:5: error: 'asm' undeclared (first use in this function)`

I then found out that this has to do with the `-std=c99` flag. I removed that flag in `lib/open62541/CMakeLists.txt`, line 347.
Then it got further with the compilation, until almost the very end where **it threw this error**:

`/home/[...]/open62541-arduino/src/arduino_opcua.cpp:9:22 fatal error: Ethernet2.h: No such file or directory`

So this is the error I have been trying to solve and failed to do so. I have tried copying the Ethernet2 library into various places where the other libraries required by that file are and trying again but I had no luck. This could have something to do with the fact that I don't really know a lot about `cmake` and `make` - maybe the library needs to be included somewhere else?

Pro, 2020-04-17:
As already mentioned, this repository is not maintained anymore. Therefore it is very likely that the build does not work anymore due to changed dependencies.
This also means that I can not help you fixing compile issues.

Sorry.

Arduino is basically just an operating system which can be flashed onto many chips. One of them is the ESP32 (https://github.com/espressif/arduino-esp32). If you are not an expert in microcontroller programming I strongly recommend that you do not invest time in my repository here as it contains complex steps to get it running built on old compiler files.

dj-fiorex, 2021-02-07:
Hi @puchm, did you built this repo?
 i'm trying to build but i have an error:
`Value for ESP32.recipe.objcopy.eep.pattern not defined`
tested with arduino 1.8.6 and 1.8.8, same error!
It seems related to the arduino compiler that search for this recipe, but platform.txt of the esp32 didn't have this recipe.

puchm, 2021-02-07:
@dj-fiorex sadly I didn't. I tried a few different things but we then decided to switch to a Raspberry Pi where we used a python version of OPC UA.
But maybe other people can help.

I am closing this since the repository is not maintained anymore, as @Pro said. If you still have the same problem, please open a new issue.

closed 2021-02-07 by puchm